realthunder/FreeCAD #650: [UI] Pad Parameters UI change

StudioPetrikas, 2022-11-30:
Hello,

There seems to be changes in the Pad pad parameters task, and it does not play well with many themes. 

[image]

If the buttons are have padding around text, the middle column becomes really narrow, preventing from reading the list of elements. 

[image]

I would propose to add these buttons below the selection window, with full, proper text, something like this:

[image]

StudioPetrikas, 2022-12-02:
The top screenshot is the default stylesheet, clean install. 

[image]

This looks like it might be a windows-only problem.

realthunder, 2022-12-02:
I see, it only happens when the edit panel can be shown in its full height.

commit 86a4293d (2022-12-02): PartDesign: fix profile editing box height

Fixes #650
@StudioPetrikas
closed 2022-12-02 by realthunder