realthunder/FreeCAD #652: Part Design:  Pad partially not filled / top-surface not covered

kotrans, 2022-12-02:
Hi
I'm working a couple of month with FreeCAD and this ist my first report.

After I attached a second Pad (red, Pad001) on top of the first pad (Pad [Face10]) half of the top of the first pad looses the covering (it is now open).

Is this a bug or do I misunderstand something?

[Test.zip](https://github.com/realthunder/FreeCAD/files/10140525/Test.zip)
[image]

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.1112.26244 +5306 (Git)
Build type: Release
Branch: LinkDaily
Hash: c64e4a49244a1c0ce9fd6455d18aa919d0db9c80
Python version: 3.10.6
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
Locale: German/Switzerland (de_CH)

realthunder, 2022-12-02:
Can you please try recompute the whole document and see if it can fix the problem (right click the document icon in the tree view, and select `Mark to recompute`). I've been seeing this kind of 'intermediate' failure quite often lately. There must be some bug hidden somewhere. I have yet to find out where. Can you recall your steps? Is it repeatable?

kotrans, 2022-12-02:
Hi Lei

Thanks you for your answer and your great job you are doing!

Mark to recompute changes nothing.

But if I enter an offset of the Sketch001 it is fixed. Try it out, it is reversible. I provided both file in the zip below.

Have a good day,

Markus
[Test with offset.zip](https://github.com/realthunder/FreeCAD/files/10141542/Test.with.offset.zip)

[image]

kotrans, 2022-12-02:
Yes, but nothing changes

realthunder, 2022-12-02:
Hmm... it works here. Can you please try again with the latest 11.28 release?

kotrans, 2022-12-02:
I downloadet the new version:

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.1128.26244 +5318 (Git)
Build type: Release
Branch: LinkDaily
Hash: d45d221edcc7a757eb4e4eb638da0db5ed2759aa
Python version: 3.10.8
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
Locale: German/Switzerland (de_CH)

It is still the same. I press right click on Test mark to recompute and then CTR+R and then F5. But nothing changes.

kotrans, 2022-12-02:
Here they are: 
[settings.zip](https://github.com/realthunder/FreeCAD/files/10142024/settings.zip)

realthunder, 2022-12-03:
Ok, now I can reproduce it. It looks like a common OCC tolerance problem. The arc you draw in the sketch is not exactly the same as the arc in the lower Pad. Adding a slight fitting in the new Pad is indeed a common way to work around the problem.

kotrans, 2022-12-03:
Thanks, it works.
I entered a fitting tolerance of 0.00001 mm.