realthunder/FreeCAD #653: settings->display->shadow: ground settings

snelweg, 2022-12-03:
When I try to use the ground settings I get an error about copying. 
The error
[image]

The bump and normal files:
[image]

The settings dialog
[image]

The version I am using right now
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.1112.26244 +5306 (Git)
Build type: Release
Branch: LinkDaily
Hash: c64e4a49244a1c0ce9fd6455d18aa919d0db9c80
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: Dutch/Netherlands (nl_NL)

snelweg, 2022-12-03:
Switched to the 28-11 release, same problem.

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.1128.26244 +5318 (Git)
Build type: Release
Branch: LinkDaily
Hash: d45d221edcc7a757eb4e4eb638da0db5ed2759aa
Python version: 3.10.8
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
Locale: Dutch/Netherlands (nl_NL)

realthunder, 2022-12-05:
Can you please try replace `/` with `\`.

snelweg, 2022-12-05:
There is some magic going on.... It won't let me change the '\'
This happens with the native file dialog setting on or off.
 
[image]

da-h, 2022-12-13:
**Possibly related**:
With the linux-AppImage, adding any image on the system results in trying to copy to the root-directory and fails with the following error:
```
 01:19:11  <Exception> Cannot copy file from /home/da-h/.FreeCAD/Gui/Stylesheets/groundtexture1.png to /groundtexture1.png
```

commit bbae415f (2022-12-13): Base: fix restoring base64 encoded file embedded in xml

Fixes #653
@snelweg
closed 2022-12-13 by realthunder