realthunder/FreeCAD #654: Crash while performing a draft Scale

lagnat, 2022-12-03:
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 2022.1128.26244 +5318 (Git)
Build type: Release
Branch: LinkDaily
Hash: d45d221edcc7a757eb4e4eb638da0db5ed2759aa
Python version: 3.10.8
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
Locale: C/Default (C)

```
Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libpython3.10.dylib           	       0x10754fc45 method_dealloc + 21
1   libpython3.10.dylib           	       0x10767135d _PyEval_EvalFrameDefault + 58973
2   libpython3.10.dylib           	       0x10754ba66 _PyFunction_Vectorcall + 582
3   libpython3.10.dylib           	       0x10755052e method_vectorcall + 158
4   libpython3.10.dylib           	       0x10766ec69 _PyEval_EvalFrameDefault + 49001
5   libpython3.10.dylib           	       0x10754ba66 _PyFunction_Vectorcall + 582
6   libpython3.10.dylib           	       0x10766ca6c _PyEval_EvalFrameDefault + 40300
7   libpython3.10.dylib           	       0x10754ba66 _PyFunction_Vectorcall + 582
8   libpython3.10.dylib           	       0x10766ca6c _PyEval_EvalFrameDefault + 40300
9   libpython3.10.dylib           	       0x10754ba66 _PyFunction_Vectorcall + 582
10  libpython3.10.dylib           	       0x10755066f method_vectorcall + 479
11  libFreeCADGui.dylib           	       0x108a0285b Gui::TaskView::TaskDialogPython::accept() + 363
12  libFreeCADGui.dylib           	       0x1089faae4 Gui::TaskView::TaskView::accept() + 100
13  libQt5Core.5.15.4.dylib       	       0x10a566600 0x10a41a000 + 1361408

```

lagnat, 2022-12-04:
The problem is with the dialog box.  Just pressing cancel is enough to cause the crash.

camelator, 2022-12-21:
I suspect an issue in the float separator:
1/ having a random object and clone it.
2/ Try to manually change the scale in the properties

=> Does not work with errors on symbols. 
I suspect something wrong with local units...

lagnat, 2022-12-25:
Clone in what workbench?

realthunder, 2023-01-11:
Should be fixed by  e3832a876d. Feel free to reopen if else.

closed 2023-01-11 by realthunder