realthunder/FreeCAD #656: Freecad-gears add-on parameters have limits.

vovodroid, 2022-12-06:
Install freecad-gears add-on via Addon Manager. Create any gear, View parameters

[image]

can't be set lower than these.

Base FreeCad allows set any value, and defaults are also different, though add-on installed in the same folder `AppData\Roaming\FreeCAD\\Mod\freecad.gears`

realthunder, 2022-12-06:
The limit setting is exposed through `Preference -> Part design -> Shape view -> Tessellation`. Change those two `lower bound` settings. The lower bound is added because a value that is too small may cause freezing when opening a document. Not sure if later OCC version still has this problem.

vovodroid, 2022-12-06:
Thanks, closing issue!