realthunder/FreeCAD #659: Addon manager freezes FreeCad

GinTonic99, 2022-12-12:
`OS: Debian GNU/Linux bookworm/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.1128.26244 +5318 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: d45d221edcc7a757eb4e4eb638da0db5ed2759aa
Python version: 3.10.8
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
Locale: German/Austria (de_AT)

trying to check for an update for asm3 WB in addon manager leads to the following error messages:

```
11:43:43  Debug: README not found at https://github.com/realthunder/FreeCAD_assembly3/blob/master/README.md
11:43:43  Traceback (most recent call last):
11:43:43    File "/tmp/.mount_freecaJO7Z0P/usr/Mod/AddonManager/addonmanager_utilities.py", line 126, in urlopen
    u = urllib2.urlopen(req, timeout=timeout)
11:43:43    File "/tmp/.mount_freecaJO7Z0P/usr/lib/python3.10/urllib/request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
11:43:43    File "/tmp/.mount_freecaJO7Z0P/usr/lib/python3.10/urllib/request.py", line 525, in open
    response = meth(req, response)
11:43:43    File "/tmp/.mount_freecaJO7Z0P/usr/lib/python3.10/urllib/request.py", line 634, in http_response
    response = self.parent.error(
11:43:44    File "/tmp/.mount_freecaJO7Z0P/usr/lib/python3.10/urllib/request.py", line 563, in error
    return self._call_chain(*args)
11:43:44    File "/tmp/.mount_freecaJO7Z0P/usr/lib/python3.10/urllib/request.py", line 496, in _call_chain
    result = func(*args)
11:43:44    File "/tmp/.mount_freecaJO7Z0P/usr/lib/python3.10/urllib/request.py", line 643, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
11:43:44  urllib.error.HTTPError: HTTP Error 404: Not Found
11:43:44  Failed to get url https://github.com/realthunder/FreeCAD_assembly3/blob/master/README.md
11:43:44  Debug: README not found at https://github.com/realthunder/FreeCAD_assembly3/blob/master/README.md
```

when trying to install a new WB (eg manipulator) the program freezes completely

the issue might have something to do with the issue reported in the forum:
[https://forum.freecadweb.org/viewtopic.php?p=643886](url)

thank you

GinTonic99, 2022-12-18:
seems to be resolved in latest snap version

closed 2022-12-18 by GinTonic99

bstott, 2022-12-23:
Please re-open --- Windows 10 64bit
OS: Windows 10 Version 2004
Word size of FreeCAD: 64-bit
Version: 2022.1128.26244 +5318 (Git)
Build type: Release
Branch: LinkDaily
Hash: d45d221edcc7a757eb4e4eb638da0db5ed2759aa
Python version: 3.10.8
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United States (en_US)

Attempting to install WB A2Plus.   Freezes downloading. Must 'Cancel'. Then, software prompts to 'Must Restart..." Upon restart the Addon is available.   But- The A2Plus addon Freezes FreeCAD.  Must kill via 'Task Manager' Windoze.

realthunder, 2022-12-23:
There is probably something wrong with the latest Windows conda build release. Please try the Libpack release instead.

bstott, 2022-12-25:
Libpack freezes addon manager on a2plus download, still. And, navigation cube gets funky when sized up from default of 132. I checked size 142. It stopped.