realthunder/FreeCAD #661: Compling LinkDaily

Nsikak1, 2022-12-13:
I tried to compile Linkdaily which turns out successful but i cant seem to use the draw style it shows this error 
"FreeCAD 0.20, Libs: 0.20R14555 (Git shallow)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

connect failed: No such file or directory
Coin info in SoGLSLShaderObject::printInfoLog(): GLSLfragmentShader  log: '0:67(1): error: no function with name 'DirectionalLight'
'
Coin error in SoGLSLShaderProgram::ensureLinking(): error: 'GL_INVALID_VALUE' (set envvar COIN_GLERROR_DEBUGGING=1 and re-run to get more information)
Coin error in SoGLSLShaderProgram::ensureLinking(): error: 'GL_INVALID_VALUE' (set envvar COIN_GLERROR_DEBUGGING=1 and re-run to get more information)
"

And  I can't find any instruction on how to go further

realthunder, 2022-12-13:
You'll need to build my fork of libCoin [here](https://github.com/realthunder/coin). If you aren't sure if your system's libCoin version is the same as my forked one, you'll have to build pivy and link agains my libCoin. See instructions [here](https://github.com/coin3d/pivy#general-build-instructions).

Nsikak1, 2022-12-13:
Thanks , would inform you if all goes well.

Nsikak1, 2022-12-14:
hi @realthunder i managed to compile your fork and build with freecad unfortunately when i try to open/create a file it says
"
connect failed: No such file or directory
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f1eafdfe520]
"
Then i tried linking with pivy as you said i compiled pivy with the attached settings 

[image]

but cant get it to link with freecad(freecad doesnt compile. it reports lots of error but this s the last part
'''
ViewProviderDatum.cpp:(.text+0xf85): undefined reference to `SoLineDetail::SoLineDetail()'
/usr/bin/ld: ViewProviderDatum.cpp:(.text+0xf9a): undefined reference to `SoLineDetail::setPartIndex(int)'
/usr/bin/ld: ViewProviderDatum.cpp:(.text+0xffa): undefined reference to `SoFaceDetail::SoFaceDetail()'
/usr/bin/ld: ViewProviderDatum.cpp:(.text+0x100f): undefined reference to `SoFaceDetail::setPartIndex(int)'
/usr/bin/ld: ViewProviderDatum.cpp:(.text+0x106c): undefined reference to `SoPointDetail::SoPointDetail()'
/usr/bin/ld: ViewProviderDatum.cpp:(.text+0x1081): undefined reference to `SoPointDetail::setCoordinateIndex(int)'
collect2: error: ld returned 1 exit status
make[2]: *** [src/Gui/CMakeFiles/FreeCADGui.dir/build.make:6535: lib/libFreeCADGui.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:6541: src/Gui/CMakeFiles/FreeCADGui.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 58%] Built target PartDesign
[ 59%] Built target Sketcher
[ 61%] Built target Robot
make: *** [Makefile:136: all] Error 2
''''

Nsikak1, 2022-12-16:
Hi, @realthunder  after so many back and forth i managed to compile your fork and install it but now i cant compile pivy maybe it has something to do with the python 3.10 installed my system?
Attached is the full compilation error
[error.txt](https://github.com/realthunder/FreeCAD/files/10245667/error.txt)

realthunder, 2022-12-19:
The error seems to be about Coin header compatibility problem. Please uninstall your system libCoin package first. And then install your local compiled one. After that, try rebuild pivy, maybe re-run cmake first.

Nsikak1, 2022-12-19:
Yo, @realthunder  it works turns out i should have uninstall system library and run "make install" it compiles, thank you. have a blessed week
[image]

closed 2022-12-19 by Nsikak1