realthunder/FreeCAD #662: Pad issue

ljungis, 2022-12-15:
[Pad_issue.zip](https://github.com/realthunder/FreeCAD/files/10237280/Pad_issue.zip)
[image]
[image]
 
How come that pads becomes holes?

realthunder, 2022-12-18:
The problem actually stems from the `Thickness` feature where OCC produced an invalid shape.

I have added a new option to validate shape on changes in `Preferences -> PartDesign -> General -> Validate shape`. It is default on. It takes effect on newly created object. For it to take effect on existing objects, mark and recompute them, and you can see where is the first shape that's having problem. `Alt + Click` the icon to try fixing it. Not all OCC problems can be fixed this way, but luckily in your case here, it can.

[image]

https://user-images.githubusercontent.com/1207888/208288844-08ee0416-1fba-4677-a0b6-e115564b3fc7.mp4

ljungis, 2022-12-18:
Thank you,

Does it ever happen that the OCC is updated?