realthunder/FreeCAD #667: TopoNaming Test

NothanUmber, 2022-12-23:
Did a small experiment where I attached a sketch to a surface which's numbering depends on a parameter.
How to reproduce:
* Load the attached design 
[PatternTest_LinkBranch.zip](https://github.com/realthunder/FreeCAD/files/10291102/PatternTest_LinkBranch.zip)
* Check out Sketch004 - it isAttached to Face7 of Loft001
* Open the "Parameters" spreadsheet and change the parameter "num_pipes" from 3 to 4
=> The error message "580.833 <App> Document.cpp(4327): Failed to recompute PatternTest_LinkBranch#Sketch004: AttachEngine3D: subshape not found Loft001.?Face7" appears - a face Loft001.?Face7 does not exist anymore

This could be easily prevented by using the XY plane with a z-offset for Sketch004 instead of binding it to a surface of the Loft. This was only a test for Topo-stuff.

Used FreeCad LinkDaily 2022.1128 (26244 +5318)

NothanUmber, 2022-12-23:
P.S.: When changing num_pipes from 3 to 4 and then back to 3 the links seem to be restored.
But when changing from 3 to 2 and then back to 3 the links stay broken.

realthunder, 2022-12-23:
> * Open the "Parameters" spreadsheet and change the parameter "num_pipes" from 3 to 4
>   => The error message "580.833  Document.cpp(4327): Failed to recompute PatternTest_LinkBranch#Sketch004: AttachEngine3D: subshape not found Loft001.?Face7" appears - a face Loft001.?Face7 does not exist anymore

This is an expected behavior. Because the face you attached does not belong to any single pattern. It is generated by combining two patterns. So when you change the pattern count, the original face is gone. There is a similar face, but obviously not the same.

BTW, the way you use patterned profiles causes Loft to produce an invalid shape. You can turn on `Split profile` option to make it right.

[image]

realthunder, 2022-12-23:
> P.S.: When changing num_pipes from 3 to 4 and then back to 3 the links seem to be restored.
> But when changing from 3 to 2 and then back to 3 the links stay broken.

There is a recent fix of losing track of topo naming change on undo/redo.

NothanUmber, 2022-12-23:
Thanks for the feedback - also thanks for the "split profile" hint!
Ok, then even with the Topo fix in place one probably shouldn't blindly use any face in sight as a reference for new sketches.

> 
> There is a recent fix of losing track of topo naming change on undo/redo.
In this case I didn't use the undo or redo feature but just changed the value back to what it was. Would of course be nice if undo and manual reversion of steps would always lead to the same internal state if possible.