realthunder/FreeCAD #670: DODO wb does not work in linkstage3

snelweg, 2022-12-26:
Drawing a couple of wires and using "insert a branch " as per documentation gives different results in 0.21 and latest Linkstage3.
Same problems with "PipeLineManager ".
It functions in 021

Linkstage3:

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.1128.26244 +5318 (Git)
Build type: Release
Branch: LinkDaily
Hash: d45d221edcc7a757eb4e4eb638da0db5ed2759aa
Python version: 3.10.8
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
Locale: Dutch/Netherlands (nl_NL)
[/code]

FreeCAD 0.21:

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.31155 (Git)
Build type: Release
Branch: master
Hash: 6f1b29769b33676edca2bd883b80bed25a0758c8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * AnimationFreeCAD 1.0.0
  * Assembly4 0.12.5
  * dodo 1.0.0
  * DynamicData 2.46.0
  * fasteners 0.4.27
  * FeedsAndSpeeds 0.5.0
  * IconThemes
  * POV-Ray-Rendering
  * Render 2022.3.0
  * sheetmetal 0.2.59
  * ShortCuts
  * TabBar
  * Woodworking 0.21.30486
[/code]

snelweg, 2023-01-08:
Nobody interested?
Seems to me that a fundamental internal part of FC linkstage is not in sync with 0.21
Might explain too why Assembly 4 misses so much of it functionality in linkstage.

realthunder, 2023-01-08:
Just happen to have missed your post. Been busy lately. I'll check soon. Also, feel free to report problem of Assembly 4, too.

snelweg, 2023-01-08:
Sorry, I didn't mean to be impatient. 
Take your time. I will make another issue for the Assembly 4 problems.

closed 2023-01-09 by realthunder

realthunder, 2023-01-09:
I've also submitted a [PR](https://github.com/oddtopus/dodo/pull/19) to dodo workbench to fix the `object still touched` problem after recompute.

commit e24dbc52 (2023-01-09): Part: fix Part.makeSweepSurface() backward compatibility

Make sure to return a single face if possible

Fixes #670