realthunder/FreeCAD #671: There are many wrong lines on the model.

KangZhaoKui, 2023-01-05:
Hi Lei, Thanks you for your great job !

When i create a model there are many wrong lines, When the mouse is hover on the model then those  lines disappear, when move the model those  lines appear again;

https://user-images.githubusercontent.com/45224709/210734052-1ba5848c-1f89-4ddf-916f-84c4a820339a.mp4

I reinstalled the windows system, still the same;

OS: Windows 10 Version 2009（Real is Windows 11,but i test it in windows10  still the same bug）
Word size of FreeCAD: 64-bit
Version: 2022.1128.26244 +5318 (Git) (FreeCAD-asm3-Daily-Win-Conda-Py3.10-20221128-x86_64)
Build type: Release
Branch: LinkDaily
Python version: 3.10.8
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
CPU:AMD Ryzen 5 1600X  
GPU:Radeon(TM) RX 460 Graphics

realthunder, 2023-01-05:
Can you please try LibPack build and see if it is ok?

KangZhaoKui, 2023-01-05:
Hi,Lei

Seems like I solved the bug with click : **Tools->Preset Configurations ->Display settings**;
I have tested many files and this bug has not appeared yet;

Before that i test lots of things , such as Preferences ->Display ->3D View ->Rendering ->Use software OpenGL (Open or close ), reinstall the windows system,  freecad releases form 2022.11.28 、2022.11.12 down to 2022.04.30 ( include LibPack build) , I almost gave up in the end !

Thank you very much ! You are doing a great job !

closed 2023-01-05 by KangZhaoKui

twu425, 2024-01-19:
I'm not sure if this is the appropriate place to put this, but I ran into this problem on the FreeCAD-Link-Stable-Win-x86_64-py3.8-20240113 build. KanZhaoKui's solution works, but I had to dig quite a bit to find it.