realthunder/FreeCAD #674: Draft_Circle generates an "Access Violation"

PaduIT, 2023-01-09:
Hello, in the   Draft workbench, the "Draft_Circle" command generates a FreeCAD crash. The   following message is displayed before the crash: "Access violation"   I noticed the   problem in the versions "Link Daily 2022.11.12" and "Link   Daily 2022.11.28"   "draft_circle"   command works perfectly in "Link Daily 2022.07.09" version   I worked with   Windows 11   

Thanks and Best regards   

Padu

YosypC, 2023-01-14:
Came here to post about this. Fortunately I'm not the only one.

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.1112.26244 +5306 (Git)
Build type: Release
Branch: LinkDaily
Hash: c64e4a49244a1c0ce9fd6455d18aa919d0db9c80
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/Europe (en_150)

realthunder, 2023-01-14:
Should be fixed by e3832a876db. Feel free to reopen if otherwise.

closed 2023-01-14 by realthunder