realthunder/FreeCAD #675: SVG import scatters objects over a wide area

julia-m-stephenson, 2023-01-10:
Importing a fairly busy SVG file into Link Daily produces this mess. 

[image]

I'm using this version
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.1128.26244 +5318 (Git)
Build type: Release
Branch: LinkDaily
Hash: d45d221edcc7a757eb4e4eb638da0db5ed2759aa
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United Kingdom (en_GB)
[/code]

Using vanilla 0.20.1 I get what I was expecting.

[image]

This is the vanilla version.

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * fasteners 0.4.27
[/code]

I've had a scan through the Vanilla FreeCAD git issues list. The only one that seems to be a possible match is this one from May last year.

[https://github.com/FreeCAD/FreeCAD/issues/6869](url)

It was fixed by this commit.

[https://github.com/FreeCAD/FreeCAD/commit/cd53c06b9899b03fe88d2a2fe8258422e7ac9c09](url)

I can't guarantee this will fix my issue but I'm pretty sure my SVG has rotates in it so may be a contender.

Let me know if you want the SVG file.

realthunder, 2023-01-10:
Can you please post the SVG file for me to verify? Thanks!

julia-m-stephenson, 2023-01-10:
[image]

Hopefully this works :-)

realthunder, 2023-01-10:
Thanks. I've cherry picked one more [commit](dc0676bb30f240e7d5f822abff365a966a2168ea) in addition to the one you found to fix the problem.

closed 2023-01-10 by realthunder