realthunder/FreeCAD #677: Additive pipe on a rotation seam path causes Invalid Spine error.

julia-m-stephenson, 2023-01-11:
I can't find anything that matches this my google-fu is weak today. This works on vanilla 0.20.1.

I've selected a sketch and a path (Rotation Seam) to make an additive pipe.

[image]

When Additive pipe is selected I get an Invalid spine error and the Rotation Seam (edge2) is de-selected. 
[image]

If I then make the rotated object visible and add the same sweep path. It works.

[image]

[image]

(Ignore the timestamps on the second two pics Spine errors only happen on the initial Additive pipe.

I have attached the test file at the point just before the additive pipe. Excuse the weird object its an example of the Mango-Jelly_Solutions YouTube course!

My version details
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.1128.26244 +5318 (Git)
Build type: Release
Branch: LinkDaily
Hash: d45d221edcc7a757eb4e4eb638da0db5ed2759aa
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United Kingdom (en_GB)

julia-m-stephenson, 2023-01-11:
[cooker_knob_link_stage.zip](https://github.com/realthunder/FreeCAD/files/10391579/cooker_knob_link_stage.zip)

julia-m-stephenson, 2023-01-11:
After some more playing you get the same error if just the sketch is selected and additional pipe is clicked.

realthunder, 2023-01-13:
There is an invalid multi-section feature. You probably added it by accident. You know when you first created the pipe, the first selection is considered as the base profile, the second is considered the sweep path, and the third and beyond are all considered as additional sections. I guess you were adding third section thinking it would belong to the sweep path. There is indeed some ambiguity here. Should be improved in the future.

Once removed it, the pipe feature can be made, but still with problem. Geometry checker will tell it is an invalid shape. You'll need to change `Corner Transition` to `Transformed` to make it work properly.

[image]

julia-m-stephenson, 2023-01-13:
Thankyou so much for taking the time to comment. Yes my mistake on adding the Edge4 in the section list. DOH!

I removed that and it worked 
[image]

It seemed corner transition was okay I didn't need to select Transformed.

The UI could be improved by having the sweep path list and the section list look the same (like the section list I think). Error handling could be improved by detecting the same object in both lists but thats probably asking a bit much!

However my first comment [https://github.com/realthunder/FreeCAD/issues/677#issue-1528833810](url) about the spine error still holds true. In vanilla FreeCAD you can select 1 Section and 1 Path and then hit the Additive pipe and it works fine. With Linkstage you still get the spine error.

Thanks again :-)

realthunder, 2023-01-13:
The problem in OP is fixed by the above commit.

Regarding the corner transition, you can use `Check geometry` to see that it's an invalid shape. I've added new feature to notify `invalid shape` in tree view.

https://user-images.githubusercontent.com/1207888/212343043-5f38b64b-e2d6-417e-b6ce-9e31b3df685e.mp4

commit 734bfcf1 (2023-01-13): PartDesign: fix Pipe creation

Fixes #677
closed 2023-01-13 by realthunder