realthunder/FreeCAD #680: Click on "Supported formats"  freezes

mgth, 2023-01-15:
On windows 11 - Win64 Conda build, on open file, I'd like to not see stl files, but when I click on "Suported formats"
The app just freezes forever, and I have to kill it.

julia-m-stephenson, 2023-01-15:
It would be helpful if you gave exact version. 

Menu
Help->About FreeCAD Linkstage Branch
[image]

Hit the button marked "Copy to clipboard" and paste it as a comment below here.

Best Regards Julia

realthunder, 2023-01-15:
If you are using `Conda` build, please try `LibPack` first.

mgth, 2023-01-20:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2023.114.26244 +5346 (Git)
Build type: Release
Branch: LinkDaily
Hash: 734bfcf15661aea45b019c0588e70e33c33ee73c
Python version: 3.10.8
Qt version: 5.15.6
Coin version: 4.0.1
OCC version: 7.6.3
Locale: French/France (fr_FR)

mgth, 2023-01-20:
Just tryed with LibPack version, but the same append.

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2023.114.26244 +5346 (Git)
Build type: Release
Branch: LinkDaily
Hash: 734bfcf15661aea45b019c0588e70e33c33ee73c
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: French/France (fr_FR)

mgth, 2023-01-20:
For information, FreeCAD 0.20 exhibits the same bug, but it seems to be fixed in 0.21.