realthunder/FreeCAD #681: Documentation Request: Installing Linkstage on Windows (for dummies)

julia-m-stephenson, 2023-01-15:
Hi,

I found the user experience installing under windows quite a challenge since solver isn't installed automatically.

So I wrote a small dummies guide for myself as I tend to forget stuff like this.

FInd a copy attached. If you like it feel free to add to GIT where appropriate and maybe update readme.md file to point to it as us Windoze users are quite dim!

Thanks
[Install new linkstage release on a Windows PC.pdf](https://github.com/realthunder/FreeCAD/files/10420256/Install.new.linkstage.release.on.a.Windows.PC.pdf)

Julia

realthunder, 2023-01-15:
Thanks for the initiative. Regarding the solver installation though, you are supposed to get prompt asking to install the solver when you open any file containing asm3 objects. The solver should then be installed and activated without requiring to restart FreeCAD. If that didn't happen to you, then there could be some problem somewhere. Are you using `Conda` build? There are user reporting that Python 3.10 seem to have problem with the last activate without restart part, so you'll still need to restart FreeCAD to make it work.

julia-m-stephenson, 2023-01-16:
Hiya,

I understand what you are saying but....

For a Windows user who is perhaps evaluating different CAD packages the experience Out Of the Box (OOB) is very poor. 
Basically you have to manually install the software and then get the solver error every time  it starts. 

This makes newbies think they made a mistake or there is a problem and look elsewhere. I fell into this trap.

A new user is unlikely to jump straight into assemblies, so the error will persist.

I don't understand why solver isn't packaged with linkstage release. 
I get that it is like that for installing Assy3 under vanilla FreeCAD, but to be honest it should install solver when Assy3 is brought in anyway. Why make users jump through hoops?

All that you wrote above is not documented for Linkstage - I spent ages trying to solve the solver issue! Hence the need for a document.

As a final whinge the naming of the libpack release suggests to me that it is an Add-on rather than a separate freestanding release using a more stable version of Python. 

As a bit of background - I had a look at FreeCAD and Linkstage nearly two years ago and thought it had promise but I was a too busy at the the time. I've now retired from a career developing embedded software and have more time to learn CAD. I really wanted to get Linkstage working as I think it has real potential so I persisted with it, these observations are based on my own experience over the last few weeks. 

I want Linkstage to succeed thats why I'm trying to make my bug reports as easy to fix as possible. 

Best Regards Julia