realthunder/FreeCAD #682: Part rendering with many extra lines issue

daproph, 2023-01-15:
Hello,

After I create a solid part and return to part design, the part will render with lines that are not part of it. The more features I add to it, the more lines I get. 
[image]
They disappear when I highlight a face but return the moment I move or rotate the part. (Hovering over the navigation cube will also cause the lines to come back.) These lines also appear when I select a face and go into sketcher. 
[image]
I was thinking that it may be my system, however I don't have this issue when I use 043022 Stable. All the versions after that do produce these lines though.

I am at a loss on how to resolve this so any help would much appreciated.
System Specs:
OS: Windows 10
CPU: Athlon X4 880K
Graphics: Radeon RX 580

 Thank you

bambukouk, 2023-01-15:
Might help if you share the file?

realthunder, 2023-01-15:
Are you using the Conda or LibPack build? Can you please also try activate `Tools -> Preset configurations -> Display settings`?

daproph, 2023-01-16:
> Are you using the Conda or LibPack build? Can you please also try activate `Tools -> Preset configurations -> Display settings`?

Using the LibPack build. Your activation suggestion worked. The parts no longer render with all the extra lines drawn. 
Thanks for your help AND thank you for all the hard work you put into fixing FreeCAD and making it something amazing Realthunder!

closed 2023-01-16 by daproph