realthunder/FreeCAD #683: 2023.01.14 crashes on macOS

michaelrommel, 2023-01-15:
Hi,

the new version does not start on macOS. The terminal shows:

```
 '/Applications/FreeCAD.app/Contents/MacOS/RunFreeCAD'
dyld[4154]: missing symbol called
/Applications/FreeCAD.app/Contents/MacOS/FreeCADLink: line 27:  4154 Abort trap: 6           "$PREFIX/bin/$name" "$@"
❯ '/Applications/FreeCAD.app/Contents/MacOS/FreeCADLink'
dyld[4162]: missing symbol called
/Applications/FreeCAD.app/Contents/MacOS/FreeCADLink: line 27:  4162 Abort trap: 6           "$PREFIX/bin/$name" "$@"
```

The previous version on this same machine runs fine.
HWInfo:
Mac Studio
64 GB
macOS: 13.1 (22C65)

  Michael.

riparz, 2023-01-16:
Same for me
[image]

dimitris-banked, 2023-01-18:
Same for M1 Max running Ventura 13.1

Zcool85, 2023-01-28:
Same for me
[image]

andyylin, 2023-02-02:
Fixed in latest release, this can now be closed.

michaelrommel, 2023-02-08:
I can confirm, that it now works for me, closing this issue - thanks a lot!!

closed 2023-02-08 by michaelrommel