realthunder/FreeCAD #684: FEM_PostFilterDataAtPoint access violation

vovodroid, 2023-01-16:
Clicking on "Select point" causes an error:

[image]

It happens in any project, sample is attached.
[PointBug.zip](https://github.com/realthunder/FreeCAD/files/10424484/PointBug.zip)

julia-m-stephenson, 2023-01-16:
Looks like there was a similar bug in Vanilla 0.19 according to this issue
[https://github.com/FreeCAD/FreeCAD/issues/7008](url)

vovodroid, 2023-01-16:
> or is it a different dialog?

I guess different. Bug you mentioned about wrong value, and my one about access violation.

julia-m-stephenson, 2023-01-16:
> > or is it a different dialog?
> 
> I guess different. Bug you mentioned about wrong value, and my one about access violation.

The second bullet point in the Issue Description says it causes an access violation in FreeCAD 0.19 

Maybe that fix didn't make it to LinkStage?

vovodroid, 2023-01-16:
> it causes an access violation in FreeCAD 0.19

It already works in 0.20

julia-m-stephenson, 2023-01-16:
> > it causes an access violation in FreeCAD 0.19
> 
> It already works in 0.20

Yes but the fix may not have made it into Linkstage yet.
It might be this issue in Vanilla Freecad 
[https://github.com/FreeCAD/FreeCAD/issues/6673](url)

I will check to see if this fix is in Linkstage
https://github.com/FreeCAD/FreeCAD/commit/e9c088c0aef2

commit a2ea9dbb (2023-01-22): Fem: selecting a point for the point result filter crashes

Fixes #684
@vovodroid
closed 2023-01-22 by realthunder