realthunder/FreeCAD #685: Additive pipe still shows spine error when just selecting section.

julia-m-stephenson, 2023-01-16:
You get the error when selecting a section and then additive pipe. (So no edge selected)
[image]
This is the profile
[image]

Example file 

[Oval_Box_and_Lid.zip](https://github.com/realthunder/FreeCAD/files/10424597/Oval_Box_and_Lid.zip)

Version details

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2023.114.26244 +5346 (Git)
Build type: Release
Branch: LinkDaily
Hash: 734bfcf15661aea45b019c0588e70e33c33ee73c
Python version: 3.10.8
Qt version: 5.15.6
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United Kingdom (en_GB)

realthunder, 2023-01-22:
Is it `Loft` or `Pipe` you are trying? Based on your screenshot, I guess it is the `Pipe`. Unlike `Loft`, the `Pipe` requires a `Sweep Path`, i.e. the `Spine`. The UI could be improve though, to not present it as an error for newly created object. Maybe later.

julia-m-stephenson, 2023-01-22:
> Is it `Loft` or `Pipe` you are trying? Based on your screenshot, I guess it is the `Pipe`. Unlike `Loft`, the `Pipe` requires a `Sweep Path`, i.e. the `Spine`. The UI could be improve though, to not present it as an error for newly created object. Maybe later.

It is Pipe. Basically there is a difference between LinkStage and Vanilla 0.20.1.

If you select the just the section and then pipe in 0.20.1 you just get a recompute error - 

[image]

Obviously when you add the sweep path / Spine this error goes away. 

This issue relates to the extra error messages seen in Linkstage. Its not a big deal. 

I'm just working through lots of examples / youtube lessons at the moment and using linkstage as my main CAD tool so if I see something that differs from 0.20.1 I'm flagging it up.