realthunder/FreeCAD #687: Empty start page on linux appimage

oficinerobotica, 2023-01-17:
The latest appimage release starts with an empty StartPage. This doesn't happen with the previous release on the same pc.

[image]

OS: KDE neon 5.26 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2023.114.26244 +5346 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 734bfcf15661aea45b019c0588e70e33c33ee73c
Python version: 3.10.8
Qt version: 5.15.6
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United States (en_US)

mwlandolsi, 2023-01-17:
Same thing on other Ubuntu 22.04 

mine is 
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 2023.114.26244 +5346 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 734bfcf15661aea45b019c0588e70e33c33ee73c
Python version: 3.10.8
Qt version: 5.15.6
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United Kingdom (en_GB)

dzervas, 2023-01-18:
I have this problem in both Wayland and X11.

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2023.114.26244 +5346 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 734bfcf15661aea45b019c0588e70e33c33ee73c
Python version: 3.10.8
Qt version: 5.15.6
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United States (en_US)
```

toastedcrumpets, 2023-01-20:
Confirmed on Arch with the snapimage

realthunder, 2023-01-23:
Can you please try [this](https://drive.google.com/file/d/14SSynSLbqomZteZ4FDFqH6bXmr46h4cN/view?usp=share_link).

dzervas, 2023-01-23:
> Can you please try [this](https://drive.google.com/file/d/14SSynSLbqomZteZ4FDFqH6bXmr46h4cN/view?usp=share_link).

Fixed for me!

oficinerobotica, 2023-01-23:
Works for me too .... should we close this report?

dzervas, 2023-01-23:
Better make a commit with the target issues that rolls the dependency back

mwlandolsi, 2023-01-23:
looks good, Thank you

oficinerobotica, 2023-02-04:
I see that the problem seems to be solved with the latest package provided by @realthunder  so I'm closing this report.
Thank you RT

closed 2023-02-04 by oficinerobotica