realthunder/FreeCAD #691: Input handled as shorcut even when selecting textbox

DanielJutz, 2023-01-21:
In the newest 2023.01 appimage release, selecting a textbox and pressing a key does not type the letter but performs the shortcut action assigned to that key.

This behaviour is not present in the previous version 2022.11.

This video shows the differing behaviour in 2022.11 and 2023.01. Trying to type "1" "2" "3" changes the view, not the text

https://user-images.githubusercontent.com/64782369/213869589-a702ac58-ca53-4196-a76e-21765e032cf6.mp4

OS: endeavouros (arch)
desktop environment: xfce (x11)

Working Version:
    FreeCAD-asm3-Daily-Conda-Py3.10-20221128-glibc2.12-x86_64.AppImage

Version with bug:
    FreeCAD-asm3-Daily-Conda-Py3.10-20230114-glibc2.12-x86_64.AppImage

realthunder, 2023-01-21:
I can't reproduce it in my computer. Can you please post your user settings here? `Tools -> Edit parameters -> Export`

DanielJutz, 2023-01-21:
[freecad_link_parameters.FCParam.zip](https://github.com/realthunder/FreeCAD/files/10472637/freecad_link_parameters.FCParam.zip) (had to zip it to be able to upload it)

I also clicked on "Reset" but after the Freecad restarted the issue is still there.

I tried it out on a separate machine:
On a fresh install of EndeavourOS the same issue appeared.
On a fresh install of Ubuntu 22.04 the issue did not appear.

akshimassar, 2023-01-21:
Hi, I have the same problem with latest AppImage. Archlinux with Awesome WM.

realthunder, 2023-01-22:
Looks like the latest qt conda package got problem on some platform. I'll make one with the old package for you to test soon.

realthunder, 2023-01-23:
Can you please try [this](https://drive.google.com/file/d/14SSynSLbqomZteZ4FDFqH6bXmr46h4cN/view?usp=share_link).

akshimassar, 2023-01-23:
2023-01-22 version works as expected for me.

DanielJutz, 2023-01-23:
> Can you please try [this](https://drive.google.com/file/d/14SSynSLbqomZteZ4FDFqH6bXmr46h4cN/view?usp=share_link).
Same for me. Input works normally with this AppImage.

evg-zhabotinsky, 2023-01-23:
Same here. Entering text in sidebar triggered shortcuts, the appimage posted here fixed the problem. Fixes the "black" start page for me too. System is Manjaro, last updated on 2022.11.08, will try to update it now and see if that helps too.

evg-zhabotinsky, 2023-01-24:
Also noticed that this appimage fixed being unable to pres `Ok` in popup windows (like constraint editing) with `Enter` key. (Haven't used FreeCad for a while, thought that was "normal".)

DanielJutz, 2023-02-03:
Issue is fixed with 2023.01.31 release. Input is handled properly.

closed 2023-02-03 by DanielJutz