realthunder/FreeCAD #693: TabBar not working.

hamza00789, 2023-01-22:
TabBar is not working in this release.  

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2023.114.26244 +5346 (Git)
Build type: Release
Branch: LinkDaily
Hash: 734bfcf15661aea45b019c0588e70e33c33ee73c
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United States (en_US) 

An error message like this appears. 

18:03:03  2.648469 <asm3.main> init_gui.py(14): no solver backend found
18:03:03  Traceback (most recent call last):
18:03:03    File "C:\Users\M.Hamza\AppData\Roaming\FreeCAD\Mod\TabBar\TabBar_Gui.py", line 574, in onPreStart
18:03:03      onStart()
18:03:03    File "C:\Users\M.Hamza\AppData\Roaming\FreeCAD\Mod\TabBar\TabBar_Gui.py", line 563, in onStart
18:03:03      onWorkbenchActivated()
18:03:03    File "C:\Users\M.Hamza\AppData\Roaming\FreeCAD\Mod\TabBar\TabBar_Gui.py", line 236, in onWorkbenchActivated
18:03:03      w = tabs()
18:03:03    File "C:\Users\M.Hamza\AppData\Roaming\FreeCAD\Mod\TabBar\TabBar_Gui.py", line 153, in tabs
18:03:03      active = Gui.activeWorkbench().__class__.__name__
18:03:03  AssertionError: No active workbench
18:03:03

hamza00789, 2023-01-24:
Is there a way to fix this in the same release? Because new release will solve this problem as usual but new release comes with new problems.

realthunder, 2023-01-24:
Try extract and copy the the following `TabBar_Gui.py` file into your TabBar installation directory (usually `C:\<username>\AppData\Romaing\FreeCAD\Mod\TabBar`) and overwrite the original one.
[TabBar_Gui.py.zip](https://github.com/realthunder/FreeCAD/files/10494326/TabBar_Gui.py.zip)

blue0ceanby, 2023-01-25:
it works, no error message
thank you

closed 2023-01-25 by hamza00789