realthunder/FreeCAD #694: appimage is broken on latest PopOS/Ubuntu,  <asm3.main> init_gui.py(14): no solver backend found

antonkulaga, 2023-01-22:
[image]
I see only black screen and 22:23:47  0.874944 <asm3.main> init_gui.py(14): no solver backend found

realthunder, 2023-01-23:
Can you please try [this](https://drive.google.com/file/d/14SSynSLbqomZteZ4FDFqH6bXmr46h4cN/view?usp=share_link).

antonkulaga, 2023-02-03:
I managed to make it work even with the previous backend by installing the solver on my system. However, black screen on the first page was a disoienting

apper002, 2023-02-03:
Should all be in the App image.

apper002, 2023-02-03:
./FreeCAD-asm3-Daily-Conda-Py3.10-20230122-glibc2.12-x86_64-RealT.AppImage
/tmp/.mount_FreeCAKguSZa
FreeCAD 0.20, Libs: 2023.122R26244 +5349 (Git)
flag
flag2
1.936934 <asm3.main> init_gui.py(14): no solver backend found

apper002, 2023-02-04:
Installing Add-On

copying macro: BCFPlugin.FCMacro
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAKguSZa/usr/Mod/AddonManager/addonmanager_utilities.py", line 126, in urlopen
    u = urllib2.urlopen(req, timeout=timeout)
  File "/tmp/.mount_FreeCAKguSZa/usr/lib/python3.10/urllib/request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
  File "/tmp/.mount_FreeCAKguSZa/usr/lib/python3.10/urllib/request.py", line 525, in open
    response = meth(req, response)
  File "/tmp/.mount_FreeCAKguSZa/usr/lib/python3.10/urllib/request.py", line 634, in http_response
    response = self.parent.error(
  File "/tmp/.mount_FreeCAKguSZa/usr/lib/python3.10/urllib/request.py", line 563, in error
    return self._call_chain(*args)
  File "/tmp/.mount_FreeCAKguSZa/usr/lib/python3.10/urllib/request.py", line 496, in _call_chain
    result = func(*args)
  File "/tmp/.mount_FreeCAKguSZa/usr/lib/python3.10/urllib/request.py", line 643, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 504: Gateway Timeout
Failed to get url https://camo.githubusercontent.com/508bd4a38bbee8e08c57f3b0af5e2604159311d31225e8a4d2f688041053ca4b/68747470733a2f2f696d672e736869656c64732e696f2f6c67746d2f67726164652f707974686f6e2f672f4475616c53504879736963732f44657369676e53504879736963732e7376673f6c6f676f3d6c67746d266c6f676f57696474683d3138
AddonManager: Debug: Error retrieving image from https://camo.githubusercontent.com/508bd4a38bbee8e08c57f3b0af5e2604159311d31225e8a4d2f688041053ca4b/68747470733a2f2f696d672e736869656c64732e696f2f6c67746d2f67726164652f707974686f6e2f672f4475616c53504879736963732f44657369676e53504879736963732e7376673f6c6f676f3d6c67746d266c6f676f57696474683d3138

apper002, 2023-02-04:
Agreed: Ubuntu is a mess nowadays.....

luzpaz, 2023-02-13:
What's left to do here ?

vido89, 2023-04-08:
Im using Mint and Im also getting an error `1.473896 <asm3.main> init_gui.py(14): no solver backend found`

vido89, 2023-04-08:
@realthunder I tried that link an same error