realthunder/FreeCAD #695: Pad with tolerance problem

akshimassar, 2023-01-23:
[image]

I'm using 2023-01-22 version, same for 2022-11-28. File in question has two failing pads.

[bathroom_organiser.FCStd.zip](https://github.com/realthunder/FreeCAD/files/10478492/bathroom_organiser.FCStd.zip)

akshimassar, 2023-01-23:
Now that is weird. I tried to determine what causes the error by deleting Pads and Pockets. Then I `Undo` the deletions, recomputed and Pad005 now works. Could you please take a look to determine what's the difference?
[bathroom_organiser_fixed_pad005.FCStd.zip](https://github.com/realthunder/FreeCAD/files/10478584/bathroom_organiser_fixed_pad005.FCStd.zip)

realthunder, 2023-01-24:
`Pad004` error is kind of expected. By reading OCC's source code, it seems offset with `Tangent` join type is not really implemented. I only included to be backward compatible with upstream.

`Pad005` problem is mostly like the same as [this](https://github.com/FreeCAD/FreeCAD/issues/8176#issuecomment-1384681552). The tolerance of `Export010` is some how corrupted.