realthunder/FreeCAD #698: External geometry Issues

hamza00789, 2023-01-25:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2023.114.26244 +5346 (Git)
Build type: Release
Branch: LinkDaily
Hash: 734bfcf15661aea45b019c0588e70e33c33ee73c
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United States (en_US)

add External geometry - add/toggle intersection geometry Both are not working properly.

hamza00789, 2023-01-25:
[image]   

The result should have been something like this.  

[image] 

[file.zip](https://github.com/realthunder/FreeCAD/files/10502844/file.zip)

hamza00789, 2023-01-25:
[image]
[image]
[image]

hamza00789, 2023-01-25:
[image]
[image]          

The result should have been something like this. 

[image]
 
[File 2.zip](https://github.com/realthunder/FreeCAD/files/10502951/File.2.zip)

hamza00789, 2023-01-25:
You may fix these issues in the next release of

hamza00789, 2023-01-25:
Even basic work in Freecad is impossible if these issues are not resolved.

hamza00789, 2023-01-26:
Can you give me some guidance on these issues?

closed 2023-01-28 by realthunder

hamza00789, 2023-01-28:
Thanks for solving the problem.  
This problem causes difficulty in working. And only you know when the new release will come.  
Can this issue be fixed in the same release as the method you described in this issue?#693

hamza00789, 2023-01-29:
Can you guide me, please?

hamza00789, 2023-01-29:
Can you guide me?

hamza00789, 2023-01-29:
please inform me.

wohltat, 2023-01-30:
@hamza00789 
Please stop spamming and let people have a weekend.

realthunder, 2023-01-30:
Yes, new builds are ready. I'll release it tomorrow. But, if you want some fix really bad, maybe try follow the build instruction and compile it on your own. You can follow the official build instruction. Just substitute the git repo.

discip, 2023-01-30:
# @realthunder
# You are an honorable man and your behavior is exemplary in the highest degree! 👍🏻

hamza00789, 2023-01-30:
If I have offended anyone because of my words, I apologize. But in my opinion, I did not say anything wrong. My native language is not English. You may find my behavior inappropriate due to the language change.

commit ba13b71a (2023-01-28): Sketcher: fix support of trimmed external BSpline edge

Fixes #698