realthunder/FreeCAD #701: Missing TreeView from View menu

szimszon, 2023-01-31:
Hi,

I wonder if you could help me.

FreeCAD version: https://github.com/realthunder/FreeCAD/releases/tag/2023.01.13-edge, Appimage 

After I removed the .FreeCAD folder from my home. I wanted to set up the UI to bring ComboView to the right and only the TreeView should be on left side. However the TreeView possibility is missing from the View menu.

[image]

Thank you!

realthunder, 2023-01-31:
You can play around with the options below. Or, try either overlay light or dark theme presets in `Tools -> Preset configurations`.

[image]

szimszon, 2023-01-31:
My faullt. Sorry. It is there. Thank you!

closed 2023-01-31 by szimszon