realthunder/FreeCAD #718: Additive / subtractive helix doesn't work or provide invalid result

akshimassar, 2023-02-13:
[image]
[helix_bug.FCStd.zip](https://github.com/realthunder/FreeCAD/files/10720193/helix_bug.FCStd.zip)

AppImage 2023-01-31. Same happens if I try to boolean my body with helix.

AlexStankov, 2023-03-29:
I am having the same issue. And looks like it is not helix's problem, issue is reproducible with other primitives - sometimes Freecad generates invalid shapes with missed face(s).

wohltat, 2023-03-31:
I tried your file and this looks like a bug.
Nevertheless i doubt that what you designed was your intention. The lower part is tilted and the upper part is vertical.
I think its better to make the lower part bigger and cut the helix off it.

akshimassar, 2023-03-31:
They should have the same axis. I don't think that subtracting instead of adding makes any difference here. It was the same for me. I somehow worked that around anyway.

wohltat, 2023-04-01:
> I don't think that subtracting instead of adding makes any difference here

I just tried it and it works:
[helix_bug_subtractive.zip](https://github.com/realthunder/FreeCAD/files/11129811/helix_bug_subtractive.zip)

Your file also had the problem that the helix needed a very specific cone angle since your base tube is conical and they did not match.

akshimassar, 2023-04-01:
I don't see why non-matching angles is a problem

wohltat, 2023-04-01:
The matching angles don't cause the buggy behavior of the broken model. 

But it was likely not what you wanted to build anyways. 
If you look closely on the helix in your file, the helix is  further outwards than the base. This is influenced by the cone angle. But even a matching cone angle doesn't fix the model.