realthunder/FreeCAD #731: Flatpak support

dsaiko, 2023-03-01:
Your branch is really really great!
Would it be possible to add flatpak support alongside snap, as snap is something very proprietary to Canonical?

dulouie, 2023-05-13:
The last release with the 'Tip' tag has flatpak support, but i run into some crashes while testing out.

realthunder, 2023-05-13:
Can you please elaborate? Also, please run in a terminal so that it will print out stack trace when it crashes. Please post the output here.

kevenwyld, 2024-01-08:
It doesn't crash for me using the latest one:

```
OS: KDE Flatpak runtime (sway/sway)
Word size of FreeCAD: 64-bit
Version: 2024.104.0.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 20240104stable)
Hash: 975453f6aa33093edbcc835760d7a65123059b98
Python 3.10.13, Qt 5.15.10, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
```

I did notice that it isn't actually published to flathub or anything though.  The workflow seems to only build the file it self.