realthunder/FreeCAD #733: FreeCAD_asm3 does not warn of keyboard shortcut reassignment conflicts

nathaneltitane, 2023-03-01:
I have just spent some time changing shortcuts itno a more logical set of combinations, and upon setting the same sequance for two unrelated commands within the sketcher selections, the preferences dialog did not prompt or warn of the conflict for this assignment.

realthunder, 2023-03-02:
It does not warn because it can now handle the same shortcut assigned to different commands, possibly in different workbench. Duplicated shortcut is listed in priority list, where you can reorder to resolve conflict, if those command may appear in the same context.

[image]

closed 2023-03-02 by realthunder

nathaneltitane, 2023-03-02:
Understood! Thank you.