realthunder/FreeCAD #734: models which are saved appear dark after reopening them from a file only with experimental rendering

kevenwyld, 2023-03-02:
I just refreshed my configs to try out some of the more modern UI features and for some reason now when i save a model, close FreeCAD, and then reopen the model it appears dark.  Here's a short clip illustrating the problem:

https://user-images.githubusercontent.com/11936535/222305339-0b3a6e33-846d-45d4-b580-6f21a9ed4b0a.mp4

I suspect I clicked some option by accident but I can't figure out what.  Any help would be appreciated.

I use the experimental rendering engine but not VBO.  If I turn the experimental rendering engine off the problem goes away.  The weird part is that if I go back to my original configuration files which have experimental rendering enabled I don't have this issue, which makes me think it's an option somewhere.

kevenwyld, 2023-03-02:
The problem goes away for the duration of a session if I select the "Shadow" draw style, and then select the "as is" draw style again.  But if I go and open the same file again in a new session the it comes back.

realthunder, 2023-03-02:
Can you please post your configuration file.

kevenwyld, 2023-03-02:
Here's my link.user.cfg and link.system.cfg .  I zipped them up and removed some personal information from a few paths.  Thanks!

[link_config_files.zip](https://github.com/realthunder/FreeCAD/files/10872654/link_config_files.zip)

kevenwyld, 2023-03-21:
Just a quick update since the snap builds are building again.  This is still an issue on:

```
OS: Ubuntu Core 20 (i3/i3)
Word size of FreeCAD: 64-bit
Version: 2023.321.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2023.03.21-edge)
Hash: 2a426c127ac4e741b11bae3d82839ee435608e1f
Python version: 3.8.10
Qt version: 5.15.7
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United States (en_US)
```

Steps to reproduce:

1. create new project
2. make any sketch
3. pad
4. save project
5. exit FreeCAD
6. open FreeCAD and the project you just saved.
7. model appears dark.
8. change view mode to "shadow"
9. change view mode back to "as is"
10. model is the default gray color again.

I'm a bit confused why this is only happening to me, but I don't see any other reports of it so that's odd.  Since it's so repeatable for me.

kevenwyld, 2023-11-24:
This seems to have resolved it self with the following release:

```
OS: Ubuntu Core 20 (sway/sway)
Word size of FreeCAD: 64-bit
Version: 2023.1102.0.38887 (Git) Snap 121
Build type: Release
Branch: tag: 20231102tip
Hash: 53783b6707f2fd2d561a02b5542f4fbe3c1eda4a
Python 3.8.10, Qt 5.15.7, Coin 4.0.1, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.0
  * Curves 0.6.14
  * fasteners 0.4.70
```

closed 2023-11-24 by kevenwyld