realthunder/FreeCAD #739: Attempting 'infinite ground' breaks render buffer

nathaneltitane, 2023-03-03:
I have been trying to replicate @officinerobotica settings and played around with the ground scale and overall 3d settings.

There are several issues arising from this:

1: setting a value around 10000000 makes the rendering break and generates artefacts all over the place, resulting in an unusable model view.

2: going back to preferences and resetting default values does not seem to have an effect of a file that is already saved (i.e.: the broken ground setup remains in the file and is not overriden by the new settings)

3: there does not seem to be any correlation to draw styles and some of the settings,, unless they are overriden by some other obscure hardcoded value somewhere?

[image]

[image]

@oficinerobotica is it possible for you to export your draw style properties only for me to have a look at?

realthunder, 2023-03-06:
The shadow render settings is saved per view. Select the `Document` icon in the tree view. You can find the property in the `View` tab.

nathaneltitane, 2023-03-12:
> 

per view, as in, within the document itself rather than as a global setting?