realthunder/FreeCAD #742: Exported .step files have no color or transparency

TJLSTW, 2023-03-06:
Hi,

I have been messing with .step export settings and somehow got it messed up, so it doesn't export any colors or transparency anymore.. Seems like it is stuck in AP203 or something.  For some reason I have a feeling that this happened after changing "Export invisible objects" ON and OFF multiple times, but I may be wrong. 
So my question is is there any "registry" I could check what type of .step it is trying to export to make sure if it's just GUIs fault or a bug of a FC. I have also tested with all available export types (AP203-AP242) and with all export settings ticked and unticked.. no luck. 
Any help would be appreciated!

SW info and some screenshots can be find at the end of this message..

Toni

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.1128.26244 +5318 (Git)
Build type: Release
Branch: LinkDaily
Hash: d45d221edcc7a757eb4e4eb638da0db5ed2759aa
Python version: 3.10.8
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United States (en_US)

[image]

[image]

[image]

[image]

realthunder, 2023-03-06:
Can you please post your model and your settings (`Tools -> Edit parameters -> Export`). You can also try `Reset` your settings and see if it backs to normal. You can later restore by `Import` the saved settings.

TJLSTW, 2023-03-06:
Hi, thank you for your reply,

Here are the program settings at the moment and a sample model. I'll try resetting later, I'm little bit busy right now.

BTW I didn't figure out how to add model here as it's not supported filetype. Had to change .FCParam to .txt too

Toni

[image]

[FC_Param_06032023.txt](https://github.com/realthunder/FreeCAD/files/10899101/FC_Param_06032023.txt)

TJLSTW, 2023-03-06:
I also tried resetting FC settings now as you suggested, no luck. Exported step is still grey.

Toni

realthunder, 2023-03-06:
I can't reproduce your problem. Can you please post your exported STEP file? Also, maybe try with the latest release.

TJLSTW, 2023-03-07:
Hi, here's the STEP. Meanwhile I also tried installing latest official version on FC and got same results, no colors. I'll try to update Link as well.

[FC_Model.step.txt](https://github.com/realthunder/FreeCAD/files/10905937/FC_Model.step.txt)

TJLSTW, 2023-03-22:
Hi, do you have any suggestions what to try next? Remove user conf. folder too and clean the windows registry etc. to remove all previous setting if there is any?

Toni

TJLSTW, 2023-04-05:
So I finally I figured it out, or at least located the issue :)

It was "Design456" folder in "C:\Users\USER\AppData\Roaming\FreeCAD\Mod" causing problems, it had to be deleted. It doesn't care if the workbench itself is installed or not but I removed it anyway as I've never actually used it. The issue returns immediately when I put pack the original "Design456" folder. Lot of python in that folder so I haven't even tried to dig deeper.

.step export and import is now working as it should, with colors and transparencies.

Toni

meprakarsa, 2024-01-02:
Hi All, any info how to solve the same problem in Mac OS?
I could not find the "Design456" folder in "Apps\(current FreeCAD)\Contents" and "Mod" inside it

[code]
OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 2023.1102.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 53783b6707f2fd2d561a02b5542f4fbe3c1eda4a
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * SearchBar 1.0.1
  * dodo 1.0.0
  * freecad.gears 1.0.0
  * InventorLoader 1.3.0
  * Render 2022.3.0
  * A2plus 0.4.60e
  * Manipulator 1.4.9
  * fasteners 0.4.24
  * Design456 0.0.1
  * sheetmetal 0.2.59
  * POV-Ray-Rendering
  * toSketch 1.0.1
  * Help 1.0.3
  * Assembly4 0.12.4
  * Assembly3 0.11.4
[/code]

Thanks
M. Ebta Prakarsa