realthunder/FreeCAD #750: Fillet/chamfer "tangent propagation"

wavexx, 2023-03-12:
When doing a fillet or chamfer operation, all continuous edges attached to the selected edge are automatically chamfered too:

If I select the central edge here:
[image]

this is what you normally get:
[image]

Is it actually possible control how this is propagated to the chained edges? Fusion360/onshape calls "tangent propagation" as part of the dress-up feature.

For example, in this case I'd actually like to chamfer the selected edge _only_.

In the chamfer operation, I naïvely tried to select the first round edge and manually set the distance to "0". This also doesn't work, irregardless of distance: apparently the operation can only use the same radius/distance on the current chain.

In this an OCCT limit or something that would be adjustable?

nathaneltitane, 2023-03-12:
@wavexx thanks for opening this ticket, was about to do so myself! Also @realthunder can we look into related fillet/chamfering issues where propagation does not occur at all or fails under certain seemingly unclear circumstances?

Thank you

nathaneltitane, 2023-03-12:
> When doing a fillet or chamfer operation, all continuous edges attached to the selected edge are automatically chamfered too:
> 
> If I select the central edge here: [image]
> 
> this is what you normally get: [image]
> 
> Is it actually possible control how this is propagated to the chained edges? Fusion360/onshape calls this "tangent propagation" as part of the dress-up feature.
> 
> For example, in this case I'd actually like to chamfer the selected edge _only_.
> 
> In the chamfer operation, I naïvely tried to select the first round edge and manually set the distance to "0". This also doesn't work, irregardless of distance: apparently the operation can only use the same radius/distance on the current chain.
> 
> In this an OCCT limit or something that would be adjustable?

in your case, does the chamfer with tangent propagation work on both sides? I get weird instances depending on the model, where one side works but the other side won't... frustrating.