realthunder/FreeCAD #751: Upstream Merge ?

wsteffe, 2023-03-13:
Hi RT, I have seen that you have tried to merge upstream with LinkDaily in the separate branch LinkMerge. 

I think that there are some problems because otherwise you would have already put the result into LinkDaily.

I am looking forward to have the merge in Linkdaily because I hope it may solve some pending issue such as, in example, the slowness of sketch solver in some cases.

realthunder, 2023-03-14:
I've been a bit busy lately. It took me more than two months to fully merge that branch. And by the time, it's already several months behind upstream, again. I just did some preliminary test, and it seems working, although I am sure there will be all sorts of problems hiding here and there. I want to spend some more time to sync with upstream again before doing a more thorough test. But if you are eager, you can try compile that branch yourself.

wsteffe, 2023-03-15:
I have cloned LinkMerge in a separate folder and tried to configure it. But cmake fails with the following error:

```
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:28 (find_package):
  Could not find a package configuration file provided by
  "Qt5WebEngineWidgets" with any of the following names:

    Qt5WebEngineWidgetsConfig.cmake
    qt5webenginewidgets-config.cmake

  Add the installation prefix of "Qt5WebEngineWidgets" to CMAKE_PREFIX_PATH
  or set "Qt5WebEngineWidgets_DIR" to a directory containing one of the above
  files.  If "Qt5WebEngineWidgets" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  cMake/FreeCAD_Helpers/SetupQt.cmake:27 (find_package)
  CMakeLists.txt:71 (include)
```
Qt5WebEngineWidgets is installed on my system (ubuntu 22.04) but  Qt5WebEngineWidgets-dev (which I think should provide on of the two files) is not available.

Hi RT, I think you are also on ubuntu 22-04. How have you solved this problem ?

wsteffe, 2023-03-15:
Sorry, libqt5webengine5 is installed, not Qt5WebEngineWidgets.

wsteffe, 2023-03-15:
No sorry again. libqt5webenginewidgets5 is also installed.
But dev version not available.

wsteffe, 2023-03-17:
Understood, it was already said in the freecad forum that the problem can be solved with:
`sudo apt install qtwebengine5-dev`

I have done and now it starts without error.

I have also opened a design made with LinkDaily.
The model tree is OK but nothing is displayed in the 3D window (aldso after making "fit all")

realthunder, 2023-03-17:
Can you send me your model with the visibility problem?

wsteffe, 2023-03-17:
It is this one:
[dielResonFilter_6_2.zip](https://github.com/realthunder/FreeCAD/files/11002528/dielResonFilter_6_2.zip)