realthunder/FreeCAD #752: Appimage fails to complains py-slvs is missing, fails to install

nathaneltitane, 2023-03-13:
py-slvs already installed through pip:

Requirement already satisfied: py-slvs in /usr/local/lib/python3.10/dist-packages (1.0.3)

appimage asks for solvespace on assembly workbench load, fails to install.

See below.

[image]

nathaneltitane, 2023-03-14:
UPDATE: Extracting the appimage and copying ps-slvs to the squashfs root lib folder fixes the warning and makes the assembly 3 workbench function as expected...

Maybe need to figure out some kind of include directive to look for it through the appimage?

Can a .deb file be generated to solve things as needed for debian based systems instead of buggin ourselves with appimage/snap?

deus-vermiculus, 2023-04-17:
I have a similar problem with the snap package. 
[image]

i have no programming experience. So i sadly have no idea what you said by "copying ps-slvs to the squashfs root lib folder" because i can not find such a folder and dont know what exactly to copy in there.

can you help me out so i can at least use my App for now? I would appreciate your help very much.

probonopd, 2023-04-21:
The AppImage _should_ bundle all required Python modules inside the AppImage and not rely on Python modules that may or may not be installed in the base system. It should be fixed.