realthunder/FreeCAD #763: Spreadsheet hangs out for a while

Pendelf, 2023-03-20:
Very long thinks when after creating a spreadsheet the first time you enter it. And if you close the tab of the spreadsheet and then open, again thinks for a long time (hangs out for a while)

Pendelf, 2023-04-18:
It seems to have something to do with the fact that I have two monitors. I recently installed Kdenlive and it is also constantly slowing down, and the program is just hanging. I tried Kdenlive on my laptop on Manjaro and there are no brakes from the word at all.

dannystaple, 2023-08-27:
I am using the latest tip release - 2023-08-11 and seeing the spreadsheet hang/spin. I am in a single monitor configuration.

Repro steps:
- Open FreeCAD
-  New project, new part.
- Enter spreadsheet workbench
- New Spreadsheet
- Enter some data, then click on another cell

It will spin at 100%, not responding.

[code]
OS: macOS 12.6.8
Word size of FreeCAD: 64-bit
Version: 2023.811.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: d31fdf6ab71bf52a96d5d3b81f3b8349bca24e27
Python 3.11.4, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.60k
  * ExplodedAssembly
[/code]

I do not have dual monitor, or Kdenlive running. I can reproduce it 100% on this build, although it may take a few clicks/moves around the spreadsheet.

dannystaple, 2023-08-27:
Note - I cannot reproduce this in the upstream 0.20.2 release (I can try a more recent/older version though)