realthunder/FreeCAD #768: File browser won't change directories - daily build/ Fedora/ KDE

linuxguy123, 2023-03-21:
User can't change directories in the file browser. 

OS: Fedora Linux 37 (Workstation Edition) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.20.31651 (Git)
Build type: Release
Branch: LinkDaily
Hash: 2a426c127ac4e741b11bae3d82839ee435608e1f
Python version: 3.11.2
Qt version: 5.15.8
Coin version: 4.0.0
OCC version: 7.6.3
Locale: English/Canada (en_CA)

Operating System: Fedora Linux 37
KDE Plasma Version: 5.27.3
KDE Frameworks Version: 5.104.0
Qt Version: 5.15.8
Kernel Version: 6.2.6-200.fc37.x86_64 (64-bit)
Graphics Platform: X11
Processors: 24 × AMD Ryzen 9 5900X 12-Core Processor
Memory: 62.7 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1060 6GB/PCIe/SSE2

[image]

Nsikak1, 2023-03-25:
can you make a video of the process from the start page to the actual problem thanks.

GinTonic99, 2023-03-26:
Workaound: press the shift key while invoking a file dialog, this should open an alternative file browser which at leat works for me