realthunder/FreeCAD #770: CfdOF cfmesh fails meshing geometry.  Geometry error ?

linuxguy123, 2023-03-22:
Steps to reproduce.

1) Install CfdOF workbench
2) Load attached example file - recalculate or not, makes no difference
3) Doubleclick Cut_Mesh under CfdAnalysis object
4) Write mesh
5) Run mesh

The mesher will hang in RT, works fine in mainline FC.  Stopping the mesh process in CfdOF will not kill the process on the computer.  That will have to be done manually. 

For some reason RT is exporting a slightly different geometry for the Cut Ojbect than the mainline version.  Tested it with Validate Shape and Fix Shape on and off, using Recompute Object after each change.

[naca2412_BH.zip](https://github.com/realthunder/FreeCAD/files/11045436/naca2412_BH.zip)

[Bad Mesh Session.txt](https://github.com/realthunder/FreeCAD/files/11045437/Bad.Mesh.Session.txt)

[Good Mesh Session.txt](https://github.com/realthunder/FreeCAD/files/11045438/Good.Mesh.Session.txt)

[image]

linuxguy123, 2023-03-24:
I confirmed there is an issue with how RT generates geometry compared to mainline FreeCAD.

The test case above runs properly in mainline FreeCAD.  If the geometry is slightly changed (make windtunnel object wider) it also runs properly in RT.

realthunder, 2023-03-26:
Which version of upstream FreeCAD you've tried? And which OpenFoam version you've installed? I tried on Linux with Fundation OpenFoam. Both mine and upstream FreeCAD got `Meshing exited with error` after click `Run mesher`.

linuxguy123, 2023-03-27:
I'll leave this issue open and reply to it when I find it happening again.   It has something to do with specific mesh parameters.