realthunder/FreeCAD #772: "Set body tip" seems broken

wohltat, 2023-03-26:
Either the "set body tip" is broken or it is completely unintuitive. 

[image]

I expect the sphere to be added to the cube and not to the cube+cone, after changing the body tip to the cube.
This is also what the tool tip suggests.

> Alt + click this icon to set this feature as the tip of the body.
> New feature will be inserted after it.

[image]

Furthermore the GUI shows inconsistent behavior. First the green label is only at the feature with that is the body tip. But sometimes every feature has those labels.
[image]

This is especially confusing since it appears seemly random at some point and it replaces the Attachment icon that also uses the Alt key.
[image]

Btw, I know that i can move the features around by dragging and with "move object after other object". The language is also inconsistent here. Why is it called object and not feature? Calling a hole or a filled in PartDesign an 'object' doesn't make much sense. 

----------------

OS: Linux Mint 20.1 (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 2023.131.26244 +5365 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 73c4ca254c5de8849d104973482e98399fb24863
Python version: 3.10.8
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United States (en_US)

realthunder, 2023-03-26:
Yes, body tip seems to be ineffective when adding primitive shapes. But it works for other features.

> This is especially confusing since it appears seemly random at some point and it replaces the Attachment icon that also uses the Alt key.

The tool tip is specific to the icon. `Alt + Click` the icon activates the specific action associated with the icon. I am not sure which part you find confusing, because the tag and the attachment are two very different icons.

> Furthermore the GUI shows inconsistent behavior. First the green label is only at the feature with that is the body tip. But sometimes every feature has those labels.

The tip is shown as a tag icon with center green, while all other tag icon are center white. The multiple color tag icons only show up if you have explicitly created a new shape/solid in the body, otherwise only the tip feature has the tag icon. Do you mean you've seen more than one tag icon with center green? 

> Btw, I know that i can move the features around by dragging and with "move object after other object". The language is also inconsistent here. Why is it called object and not feature? Calling a hole or a filled in PartDesign an 'object' doesn't make much sense.

Yes, I agree. But object and feature are really the same thing in FreeCAD, in a much deeper sense. For example, the `Hole` feature actually carries all shape geometry from all of its ancestor features. It is very hard to infer which of the geometry belongs to which ancestor feature.

BloodyRain2k, 2023-04-09:
I agree on the confusing wording for "moving" a feature, because we're actually just reordering them.

There *is* somewhere the function to actually move something from one part to another part, but I haven't seen it in a long time. Small possiblity that it's a Main branch feature, which I haven't used in over a year, so that would explain that.
Anyways, the wording for this function is almost the exact same, but it's doing something completely different.

So I would second a rewording for the feature reordering entry. "Move object after object" sounds confusing to begin with, even when you know the context and what it does...
Maybe "Reorder feature" ?

And regarding the "Set tip" function do I also have something to add: it is not showing up for "Base" features.
[image]
The Part's tip can be manually set to it through the property panel, but I find it weird that the entry is missing here.