realthunder/FreeCAD #776: Suggested feature: Let FreeCAD provide choices of GUI colors with good contrast to make all elements visible

raindropsfromsky, 2023-03-29:
The **Edit > Preferences...** menu of FreeCAd allows us to select colors for toolbar areas and "background" in the main panel.

However, this flexibility is not of much use to the user, because in most cases the GUI becomes unreadable.

In some cases, the toolbar buttons become difficult to see because the contrast is extremely low.
In other cases, the chosen "background" color does not allow us to see the constraints and/or the selection rectangle and/or the mouse cursor.

For example, I tried to match the background color to Blender. But now I am unable to see the tool icon, the coordinates and the constraint, because all of them have very low contrast with the background color.
[image]

This also happened when I tried to keep the background color to white (as in  OnShape).
[image]
In this case, the contrast is relatively better, but I cannot see the tool icon, and the mouse cursor.
Worse, the lines of the elements become invisible.

For example, there are a couple of lines and a arc here:
[image]

In this example, there is a line:
[image]
Also, if we draw a rectangle with the mouse to select multiple items, the rectangle itself remains invisible.
It is very difficult to select items this way.

Now it is a tough task for the user to search for multiple color controls and change their setting to make them easily readable.
A new user does not even know which control will serve his purpose.
It becomes a huge trial-and-error exercise!

Such unforeseen problems would frustrate a novice user.
Changing the theme should not be so hard!

**Desired:**
FreeCAD should provide themes in which all colors are well-chosen to offer a good contrast.

A User may choose a light or dark theme to suit his ambient lighting condition.
But once he chooses any theme, it should not hide some elements of the display or make it difficult to read.

wohltat, 2023-03-29:
This bothers me almost every time when I create a new model. 
When viewing a flat object from the top then it is white and the lines cannot be seen because the light reflexion is so bright.

What helps as a work around for me was, setting the default shape color to something darker in settings, part design, shape appearance, shape color.

Would be nice if this was default out of the box.

raindropsfromsky, 2023-03-29:
Thanks for the response. I tried the presets, but didn't find them suited for me. 

It would be good if a large number of user-contributed themes are made available on the forum, with thumbnails.
The user can look at the thumbnails, and download the theme he likes.

It is also a good strategy to create themes that copy the appearances of all well-known CAD and 3D modeling software, because many users are already used to those GUIs. They would love to just adjust the FreeCAD GUI to their favorite software and continue.

Let him avoid the customization altogether. 

A first-time user should be able to try his hand at modeling, which is the real exciting part.
But most users would struggle to change the GUI's looks, and lose all the fun.