realthunder/FreeCAD #782: Allow referencing of same sketch's constraints

YosypC, 2023-04-03:
I don't quite understand the inner workings of FC, but whenever I name a constraint like a length and I try referencing it in the same sketch, I get a "Reference_ constraint from this sketch cannot be used in this expression.". This is a total bummer for me, since I either have to create a spreadsheet or reference, if possible, another original measurement.
Tell me if you all need additional explanations.

`OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2023.131.26244 +5365 (Git)
Build type: Release
Branch: LinkDaily
Hash: 73c4ca254c5de8849d104973482e98399fb24863
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/Europe (en_150)
`

BloodyRain2k, 2023-04-09:
I can understand why this is an annoying issue, I've been there often enough myself.
The reason why it's not allowed (yet) is because it could easily lead to cyclic dependancy.

For edges, the easiest work around is the "Equal" constraint.
If you need an offset from an edge for example, just create a construction line that spans the gap, constrain it with the intended length. Then just create another construction line in a different place where this offset is also needed and link the two construction lines with the "Equal" constraint. Now you can affect multiple constraints by changing the value of one constraint.

It works the same way for angles, but it's more complex, as you need more than just a single construction line.
From the top of my head, it'd be a min. of 3 lines: two from the "center" of the arc, to define the radius and angle between, with the 3rd being from end to end of the first two lines. Then that setup can be placed else where and after constraining all edges with there respective original versions, it should copy the angle.

But yes, it would be nice if the Sketcher could check the referenced constraint first and only reject expression values, not fixed values. If I have an arc or distance that's directly set in the sketch, it should be referenceable. It should only reject evaluated values, as these could potentially lead to infinite recalculations.

YosypC, 2023-04-09:
I needed an edge to be 1/3 of another on the same sketch, so I immediately resorted in referencing it and the problem occurred. Obviously I can't make it equal.

I don't see an issue tho': if a cyclic dependency is found, don't allow the last unit to be used. But normally it shouldn't be a limitation.

YosypC, 2023-05-01:
I tried once again in a completely different project and I was able to make it work inside not only one, but two sketches.
[image]

I have no idea how I got the initial message, I am so confused.

BloodyRain2k, 2023-05-03:
I recently ran into this myself again.
The confusing part is, that a defined constraint is fine (entered or calculated), but a measuring (reference) constraint is not, for whatever internal reason.

closed 2026-01-14 by YosypC