realthunder/FreeCAD #783: Feature request: Make a powerful array creation system in sketcher

kaushikbm, 2023-04-09:
Currently in **Sketcher** > 
[image]
**Create array** is having only basic array function. 
[image]
Please add...
- Individual / linked X & Y distance values and angles in this dialog. 
- Alternate line X & Y offset distance and angles for making zigzag patterns etc.

Also please add another icon with dialog box for a polar array with complete control like...
- Center reference selection/last selected vertex
- Number of copies/clones
- Distribution angle (default 360)
- Automatic calculation & manual entry of angle between copies/clones

Maybe more can be added.
I think this will be a great enhancement and will make life easier.

Pendelf, 2023-04-09:
It seems to me that all tools of the type lack the function of selecting elements. When you don’t need all the elements you’ve chosen. As an example we can take a circular array, let’s say we copy five elements by 360 degrees, but let’s say we keep only the first, the second and the last, respectively we choose them and leave them, and the rest after pressing the apply button disappear. Similarly to other sets types

YosypC, 2023-04-09:
I desperately all the described above functions too.

BloodyRain2k, 2023-04-10:
This could be made as a "proof of concept" as a macro, since they can also use Qt for UIs.
It doesn't sound to me like this needs new core functions, just automation of a couple of manual processes, which is exactly what macros are for.

kaushikbm, 2023-04-10:
I'd like to say that modifying the core functions will make the software more rugged and user friendly like other professional CAD systems. Even the UI can be overhauled and made simpler and cleaner with less icons.

BloodyRain2k, 2023-04-10:
No arguing there, but modifying the core is a lot of work because you have to ensure that your changes didn't break something else.

And macros won't break anything (maybe crash your session at worst). So they'd be ideal to get the functionality in, and once it's being used, it will quickly show what needs to be changed to make it better, which can be done fast with macros.
Making changes to a macro takes seconds. Recompiling FreeCAD might take minutes. Even just restarting it already takes longer than making changes to a macro and re-running it. So in this case, it's non-sensical to try a core implementation *first*.

Once the macro and it's usecases have matured, there's a good foundation from where to derive what needs to be in place to make it a core functionality by upgrading the core functions used.

kaushikbm, 2023-04-10:
Totally agree with your suggestion and implementation route. I'd like to see what @realthunder has to say :)

YosypC, 2023-04-10:
> No arguing there, but modifying the core is a lot of work because you have to ensure that your changes didn't break something else.
>
> And macros won't break anything (maybe crash your session at [...]

Macros can be a good starting point but their results cannot be changed as far as I know. By having a parametric function already compiled, we can make changes on the go without deleting the macro geometry and applying it again. Macros can be a workaround and not a solution in many cases.