realthunder/FreeCAD #788: "Display -> Draw styles" settings do not update currently opened projects

YosypC, 2023-04-09:
I've found a [RT's post](https://forum.freecad.org/viewtopic.php?style=5&t=45780&start=10#p394929) on the forum in which he changes the Draw styles dynamically. I can only apply the new ones on new documents.

In relation to [this](https://forum.freecad.org/viewtopic.php?style=5&p=674509#p674509) forum post.

`OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2023.131.26244 +5365 (Git)
Build type: Release
Branch: LinkDaily
Hash: 73c4ca254c5de8849d104973482e98399fb24863
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/Europe (en_150)`

realthunder, 2023-04-09:
You can change the draw style of an existing document through document's view property. Select the document icon in the tree view, and access the property through property view. I'll make change to Preference dialog to add the 'Apply' button for existing document.

YosypC, 2023-04-10:
Some properties are global and only accessible from the settings menu, as "Line width multiplier" and similar (hence, they update the document automatically): these options are listed along with document side ones without a clear distinction.