realthunder/FreeCAD #790: RefineShape of Part Workbench not working

lilaL, 2023-04-10:
[RefineShape](https://wiki.freecad.org/Part_RefineShape) works as expected in:
```
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
```
certain edges and faces cleaned up after this operation but not in the current Link Branch:

```
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 2023.131.26244 +5365 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 73c4ca254c5de8849d104973482e98399fb24863
Python version: 3.10.8
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
```

realthunder, 2023-04-10:
Can you please post a model showing the problem?

lilaL, 2023-04-14:
sure,
[testRefineShape.zip](https://github.com/realthunder/FreeCAD/files/11236046/testRefineShape.zip)

northcoastam, 2023-04-17:
Came here to check for this bug.  I can confirm the same behaviour with two stl files I just tested.  FreeCAD 0.20.1 cleans up the planar triangles nicely, but 2023.131.26244 +5365 (Git) Win 64 Libpack makes no visible changes.

northcoastam, 2023-07-30:
Just tested with "Release 20230729tip" and appears to be working as expected.