realthunder/FreeCAD #791: How about a new release ?

linuxguy123, 2023-04-13:
I'm loving 2023-01-31.   Any chance we could get a new release ?

Pendelf, 2023-04-13:
Also waiting for a new release, but as far as I know realthunder is now busy implementing topological problem solution in the main branch of freecad. Plus I wonder on which version of freecad the new version will be based, if I do not confuse the current version linkstage3, based on version 0.19 and already now in the current version 0.20 there are features that are not in the fork rialthunder. And in general, it seems that there is no welcome discussion such as this, such things should be done on the forum, and here to report problems and bugs

realthunder, 2023-04-13:
I am actually busy setting up github auto build for quite some time. It is mostly working now. There will be more frequent release very soon, as well as big merge release, which I haven't got time to seriously test yet.

thx1111, 2023-05-21:
I'm not sure the best place to raise this question, but - What are the thoughts or plans about the relationship between the linkstage3 fork, the upcoming merge, and the Assembly3 and Assembly4 workbenches?  Will Assembly3 continue to be supported?  Should we all be moving to Assembly4 now, when using linkstage3?  If Assembly3 is to be considered "obsolete", then do we want to continue submitting bug reports for Assembly3?

realthunder, 2023-05-21:
LinkStage3 is obsolete. I will maintain LinkDaily in parallel with LinkMerge. The later is for testing merge with upstream. The once found stable I'll merge it to LinkDaily.

Assembly3 is still supported. I plan to either find another solver or write one myself, but not in immediate future.

thx1111, 2023-05-21:
So, Assembly4 is not something based upon or derived from your work that is meant to solve the same original issues?  And vice versa, you don't plan to necessarily use Assembly4 as a framework for any future solver you would develop?

realthunder, 2023-05-21:
Assebmly4 is developed by another developer. It takes a different approach which places components with user defined local coordinate systems, hence does not require a solver to work. Some may find its workflow more natural, but others don't.