realthunder/FreeCAD #793: Compiling requirements (LibPack, Python...)

YosypC, 2023-04-16:
I'm attempting to compile the latest Linkstage3 but I'm having some issues.
To narrow it down, I first have to know which [LibPack](https://github.com/FreeCAD/FreeCAD-LibPack) to use with the latest iteration, and whether or not some additional software is required or simply optional. The [Wiki compiling page](https://wiki.freecad.org/Compile_on_Windows) obviously isn't meant for Linkstage3, and the old [Assembly3 wiki](https://github.com/realthunder/FreeCAD_assembly3/wiki) is not up to date.

I checked the readme in the GH source, but couldn't find anything.

realthunder, 2023-04-16:
`LinkStage3` branch is not maintained anymore. I suggest you try the `LinkDaily` branch. I am building it using LibPack 12.6.1 in  https://github.com/realthunder/FreeCAD/releases/tag/libpack. My cmake configuration is as follows,

```
        cmake -G "Visual Studio 16 2019" -A x64 ^
            -DBUILD_ENABLE_CXX_STD:STRING=C++17 ^
            -DFREECAD_LIBPACK_DIR=%libpackpath% ^
            -DFREECAD_USE_PYBIND11:BOOL=ON ^
            -DBUILD_FLAT_MESH:BOOL=ON ^
            -DBUILD_FEM_NETGEN:BOOL=ON ^
            -DOCC_INCLUDE_DIR=%libpackpath%include/opencascade ^
            -DPYTHON_EXECUTABLE=%libpackpath%/bin/python.exe ^
            ..
```

YosypC, 2023-04-20:
I've started the VS debug build and got this:
```
Exception thrown: read access violation.
**_Pnext** was 0xFFFFFFFFFFFFFFFF.
```

[image]

realthunder, 2023-04-25:
I don't work on Windows, so a bit unfamiliar there, especially the debug build. I have never done a proper debug FreeCAD build. The Qt symbol is not distributed with the LibPack (I didn't build them), but that shouldn't affect debugging unless you are dealing with Qt related bugs. The most troublesome part is Python modules. You'll need to have debug version of all kinds of Python modules installed. Anyway, if you aren't a developer, I suggest you try the release build only. Or, maybe ask around in the FreeCAD forum on how to debug on Windows.

closed 2026-01-13 by YosypC