realthunder/FreeCAD #794: Issues compiling on Windows 10

YosypC, 2023-04-17:
Attempting to compile "691ee74 - Renderer: fix outline drawing" for the first time. Getting some (hopefully unimportant) errors, and 

<details>

<summary>Debug output</summary>

``` CMake Deprecation Warning at CMakeLists.txt:8 (cmake_policy):
  The OLD behavior for policy CMP0072 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.19045.
   Entering             C:/Users/Yosyp/Documents/Git/source/repos/FreeCAD/cMake/FreeCAD_Helpers
   Returning to         C:/Users/Yosyp/Documents/Git/source/repos/FreeCAD
Compiler: MSVC, version: 19.35.32217.1
prefix: C:\Utilities\FreeCAD
bindir: bin
datadir: data
docdir: doc
includedir: include
libdir: lib
cmake: 3.26.3
PyCXX found:
  Headers:  C:/Users/Yosyp/Documents/Git/source/repos/FreeCAD/src
  Sources:  C:/Users/Yosyp/Documents/Git/source/repos/FreeCAD/src/CXX
  Version:  6.2.8
Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
=====================================================
Doxygen not found, will not build documentation.     
=====================================================

PyCXX found:
  Headers:  C:/Users/Yosyp/Documents/Git/source/repos/FreeCAD/src
  Sources:  C:/Users/Yosyp/Documents/Git/source/repos/FreeCAD/src/CXX
  Version:  6.2.8
CMake Warning (dev) at C:/Utilities/cmake-3.26.3-windows-x86_64/share/cmake-3.26/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (OCC) does
  not match the name of the calling package (OpenCasCade).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cMake/FindOpenCasCade.cmake:120 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  cMake/FreeCAD_Helpers/SetupOpenCasCade.cmake:4 (find_package)
  CMakeLists.txt:51 (SetupOpenCasCade)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OCE/OpenCASCADE version: 7.6.0
-- OCE/OpenCASCADE include directory: C:/Utilities/FreeCADLibs_12.6.1_x64_VC17/inc
-- OCE/OpenCASCADE shared libraries directory: C:/Utilities/FreeCADLibs_12.6.1_x64_VC17/win64/vc14/lib
VTK components: vtkCommonCore;vtkCommonDataModel;vtkFiltersVerdict;vtkIOXML;vtkFiltersCore;vtkFiltersGeneral;vtkIOLegacy;vtkFiltersExtraction;vtkFiltersSources;vtkFiltersGeometry;vtkhdf5;vtkRenderingCore;vtkInteractionStyle;vtkRenderingFreeType;vtkRenderingOpenGL2
Check for medfile (libmed and libmedc) ...
Boost toolset is unknown (compiler MSVC 19.35.32217.1)
Boost toolset is unknown (compiler MSVC 19.35.32217.1)
Boost toolset is unknown (compiler MSVC 19.35.32217.1)
Boost toolset is unknown (compiler MSVC 19.35.32217.1)
Boost toolset is unknown (compiler MSVC 19.35.32217.1)
Boost toolset is unknown (compiler MSVC 19.35.32217.1)
Boost toolset is unknown (compiler MSVC 19.35.32217.1)
Boost toolset is unknown (compiler MSVC 19.35.32217.1)
-- NOTE: Boost version 1.78.0 >= 1.78 and building on Windows: (1) disabling Win32 Regex Localization (2) enabling generic cmath.
Found Netgen: C:/Utilities/FreeCADLibs_12.6.1_x64_VC17/cmake
Found NETGEN version 6.2, calculated: 395833
CMake Warning (dev) at C:/Utilities/cmake-3.26.3-windows-x86_64/share/cmake-3.26/Modules/FindFreetype.cmake:171 (mark_as_advanced):
  Policy CMP0102 is not set: The variable named
  "FREETYPE_INCLUDE_DIR_freetype2" is not in the cache.  This results in an
  empty cache entry which is no longer created when policy CMP0102 is set to
[... 285 lines cut ...]
  CMakeLists.txt:76 (SetupCoin3D)
This warning is for project developers.  Use -Wno-dev to suppress it.

Coin3D libraries found
Shiboken2Config: Using default python: .cp38-win_amd64
SHIBOKEN_PYTHON_INCLUDE_DIRS computed to value: 'C:/Utilities/FreeCADLibs_12.6.1_x64_VC17/bin/include'
SHIBOKEN_PYTHON_LIBRARIES computed to value: ''
libshiboken built for Release
===============================================
PySide2 Python module found at C:\Utilities\FreeCADLibs_12.6.1_x64_VC17\bin\lib\site-packages\PySide2.
===============================================

Found Matplotlib: C:/Utilities/FreeCADLibs_12.6.1_x64_VC17/bin/lib/site-packages/matplotlib (found version "3.3.3")
Platform is 64-bit, set -D_OCC64
Performing Test _flag_found
Performing Test _flag_found - Failed
Performing Test _flag_found
Performing Test _flag_found - Failed
CMake Error at src/Mod/Assembly/Gui/CMakeLists.txt:35 (fc_wrap_cpp):
  Unknown CMake command "fc_wrap_cpp".

Configuring incomplete, errors occurred!
```

</details>

realthunder, 2023-04-17:
Did you explicitly set `BUILD_ASSEMBLY=On` when doing cmake configure? The built-in Assembly module of FreeCAD is deprecated. Please do not enable it.

YosypC, 2023-04-18:
> Did you explicitly set `BUILD_ASSEMBLY=On` when doing cmake configure? The built-in Assembly module of FreeCAD is deprecated. Please do not enable it.

That was the problem. Now corrected.

closed 2023-04-18 by YosypC