realthunder/FreeCAD #797: Can only reference one of out 3 bodies of a file

YosypC, 2023-04-25:
Trying some multiple files workflow. I need to reference a body's sketch from a pad in a body situated into another file, in the same folder, but it doesn't let me reference the body nor the other body.
I eventually found out I can reference the sketch from the document itself instead of navigating into the body "containing" it, but it's still an issue.

[image]
[image]

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2023.131.26244 +5365 (Git)
Build type: Release
Branch: LinkDaily
Hash: 73c4ca254c5de8849d104973482e98399fb24863
Python version: 3.8.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/Europe (en_150)
```

realthunder, 2023-04-25:
If there is no circular reference, you should be able to reference an object in an external document. Can you elaborate? Where is the sketch, in the `Master`? And where do you want to reference it from.

YosypC, 2023-04-26:
I apologize for my idiocy (because I already knew how this worked), I think I tried referencing the objects with labels using the internal name reference syntax and vice versa, that is why it worked only for one body, although I am not fully sure since I was very confused yesterday and didn't memorize all the attempts made. Also:

> Unfortunately, the integrated checker sometimes claims that a valid name doesn't exist. Continue typing anyway.

My ignorance, paired with an apparent issue with the checker as indicated in the [wiki](https://wiki.freecad.org/Expressions#Cross-document_linking), didn't allow me to continue.

I'd like to keep the issue open since there's apparently a problem, although as of right now I can successfully reference any body in **Components**, hence I'll close it and maybe open it again with specific files able to replicate the problem fully and constantly.

For newbies like me: in FC scripts, **<\<Object>\>** references the _Label_ property of a body (what's seen in the tree), while a simple **Object** references its internal name which generally cannot be changed and it's normally hidden by the user, as explained in the Wiki page above. You also need the Hash sign **"#"** after calling another file to reference its contents.

As always, sorry @realthunder for wasting your time.

closed 2023-04-26 by YosypC