realthunder/FreeCAD #806: unicode

yahbluez, 2023-05-07:
The feature to enter unicode characters got lost in one of the last updates.

ctrl-shift numbers did not longer work but did it some weeks before.

I have files using unicodes and if  i open the expression dialog i see the unicode characters but can't type in new ones and also got an error message: "syntax error, unexpected UNIT"

[image]

realthunder, 2023-05-13:
Please post the `About` information

yahbluez, 2023-05-14:
OS: Ubuntu Core 20 (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 2023.321.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2023.03.21-edge)
Hash: 2a426c127ac4e741b11bae3d82839ee435608e1f
Python version: 3.8.10
Qt version: 5.15.7
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United States (en_US)

realthunder, 2023-05-20:
Can you please try the new `Tip` release?

yahbluez, 2023-05-20:
Tested appimage and snap tip-version today, same bug.
If i open an existing freecad file where some inputs use unicode, 
i see the code but got the error message and can not close the input without deleting the unicode data.

yahbluez, 2023-05-21:
There is a similar bug in the vanilla version too, i guess this two bugs have some common:

https://github.com/FreeCAD/FreeCAD/issues/9365

My guess is that the location of this bug is somewhere in the common codebase.

realthunder, 2023-05-21:
How you are going to use the expression? Can you please post a file showing the problem? If it is used as a string, you should quote the string, e.g. `<<⌀>>`

yahbluez, 2023-05-21:
Steps to demonstrate the bug:
- new file
- draft WB
- Shape from Text
- Try to enter a unicode character => did not work anymore, worked weeks ago

In linux i type: ctrl-shift-u and than the number of the unicode character: ♥
This was: ctrl-shift-u-2665 and makes the heart symbol.

This worked as expected in freecad and stopped working around a month or longer ago.

Because it happens in your branch and the vanilla version, i guess it is something in the shared code part.

Roy-043, 2023-05-22:
> Steps to demonstrate the bug:

I experience no problems with this scenario:
https://github.com/FreeCAD/FreeCAD/issues/9365#issuecomment-1557473587